NVIDIA-AI-IOT/cuDLA-samples #6: Protobuf error when build_dla_standalone_loadable.sh

chenscottus, 2023-10-04:
Hello,

Protobuf error when build_dla_standalone_loadable.sh in Jetpack 5.1.2 in Orin 32GB:

  [10/03/2023-16:57:21] [I] Start parsing network model
[libprotobuf ERROR google/protobuf/text_format.cc:298] Error parsing text-format onnx2trt_onnx.ModelProto: 1:9: Message type "onnx2trt_onnx.ModelProto" has no field named "version".
[10/03/2023-16:57:21] [E] [TRT] ModelImporter.cpp:688: Failed to parse ONNX model from file: data/model/yolov5_trimmed_qat.onnx
[10/03/2023-16:57:21] [E] Failed to parse onnx file

Thanks!

-Scott

lynettez, 2023-10-11:
Hi @chenscottus, sorry for the late reply. looks like the onnx model is invalid, could you try to fetch the model from git-lfs and build again?
git lfs fetch --all

2yjia, 2024-01-19:
nvidia@nvidia-desktop:/usr/src/tensorrt$ git apply --verbose trtexec-dla-standalone-trtv8.5.patch
Checking patch samples/common/sampleEngines.cpp...
error: while searching for:
        setLayerOutputTypes(network, build.layerOutputTypes);
    }

    if (build.safe)
    {
        config.setEngineCapability(sys.DLACore != -1 ? EngineCapability::kDLA_STANDALONE : EngineCapability::kSAFETY);
    }

    if (build.restricted)

error: patch failed: samples/common/sampleEngines.cpp:968
error: samples/common/sampleEngines.cpp: patch does not apply
Checking patch samples/common/sampleOptions.cpp...
error: while searching for:
    getAndDelOption(arguments, "--fp16", fp16);
    getAndDelOption(arguments, "--int8", int8);
    getAndDelOption(arguments, "--safe", safe);
    getAndDelOption(arguments, "--consistency", consistency);
    getAndDelOption(arguments, "--restricted", restricted);
    getAndDelOption(arguments, "--buildOnly", buildOnly);

error: patch failed: samples/common/sampleOptions.cpp:844
error: samples/common/sampleOptions.cpp: patch does not apply
Checking patch samples/common/sampleOptions.h...
error: while searching for:
    LayerPrecisions layerPrecisions;
    LayerOutputTypes layerOutputTypes;
    bool safe{false};
    bool consistency{false};
    bool restricted{false};
    bool buildOnly{false};

error: patch failed: samples/common/sampleOptions.h:147
error: samples/common/sampleOptions.h: patch does not apply
@lynettez

lynettez, 2024-02-01:
No reply in 2 months. closing

closed 2024-02-01 by lynettez