NVIDIA-AI-IOT/cuDLA-samples #7: Standalone mode and performance issue

tanfengji-monarchtractor, 2023-10-06:
Hi, I'm using jetpack 5.1.2. The hybrid mode works fine, but the standalone mode returns the error below:
[image]
How may I fix this?

Besides, the performance is for hybrid mode is below:
[image]
The MAP is 0.354 instead of 0.37+ mentioned in the README. Any possible reasons? Many thanks

zerollzeng, 2023-11-16:
Sorry for the late response, please pull the latest code and rebuild with `make validate_cudla_int8 USE_DLA_STANDALONE_MODE=1 USE_DETERMINISTIC_SEMAPHORE=1`

Root cause in https://github.com/NVIDIA-AI-IOT/cuDLA-samples/issues/5

ou525, 2024-01-26:
[image]
The latest code pulled, according to the above command, new problems appeared

zerollzeng, 2024-01-27:
Which JP/DOS version you are using?

ou525, 2024-01-29:
[image]
@zerollzeng

ou525, 2024-01-29:
[image]
@zerollzeng

ou525, 2024-01-29:
[image]
@zerollzeng

ou525, 2024-01-29:
[image]
@zerollzeng

zerollzeng, 2024-02-01:
I didn't test on JP 5.1. but JP 6.0 should work, could you please upgrade to it?

+ @lynettez

ou525, 2024-02-02:
By replacing the nvsci header file with version 6.0, I have successfully compiled and run

zerollzeng, 2024-02-07:
Glad you solve it, I'm closing this.

closed 2024-02-07 by zerollzeng

HelloChang1, 2024-04-07:
> By replacing the nvsci header file with version 6.0, I have successfully compiled and run

hai，where can I find the the nvsci header file with version 6.0?