NVIDIA-AI-IOT/cuDLA-samples #12: fatal error: nvscibuf.h: No such file or directory

WangFengtu1996, 2024-01-18:
* Env
```
(base) orin@orin-root:~/workspace/cuDLA-samples$ sudo jetson_release
[sudo] password for orin:
Software part of jetson-stats 4.2.4 - (c) 2024, Raffaello Bonghi
Model: Jetson AGX Orin Developer Kit - Jetpack 5.1.2 [L4T 35.4.1]
NV Power Mode[0]: MAXN
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - P-Number: p3701-0005
 - Module: NVIDIA Jetson AGX Orin (64GB ram)
Platform:
 - Distribution: Ubuntu 20.04 focal
 - Release: 5.10.120-tegra
jtop:
 - Version: 4.2.4
 - Service: Active
Libraries:
 - CUDA: 11.4.315
 - cuDNN: 8.6.0.166
 - TensorRT: 5.1.2
 - VPI: 2.3.9
 - Vulkan: 1.3.204
 - OpenCV: 4.6.0 - with CUDA: YES
```
* build command
I try both in hybrid mode and standalone mode. I got same error. `src/cudla_context_standalone.h:30:10: fatal error: nvscibuf.h: No such file or directory`
```
make run 
make run USE_DLA_STANDALONE_MODE=1
```

* I search solution  in Internel. But I can not fix the bug . https://forums.developer.nvidia.com/t/where-is-nvscibuf-h/107802

2yjia, 2024-01-19:
Same problem, have you solved it?

WangFengtu1996, 2024-01-19:
NO. I think may need update jetpack.
I want to use DLA HW by TensorRT. I read the code. It is not using TensorRT to inference model.

2yjia, 2024-01-19:
I solved this problem.
Refer to this URL：https://forums.developer.nvidia.com/t/where-are-the-nvsci-headers/221091
Download the BSP corresponding to jetpack. 
The corresponding header files contained in public_sources.tbz2.
Modify the makefile and add the path

WangFengtu1996, 2024-01-22:
About the library nvscibuf nvscisync, can modify Makefile,add library path
```
-L /usr/lib/aarch64-linux-gnu/tegra -l nvscibuf -l nvscisync
```

closed 2024-01-22 by WangFengtu1996