NVIDIA-AI-IOT/cuDLA-samples #16: Size of input tensor in ReformatImage function

dattv, 2024-01-22:
Hi all.

I'm wondering that input resolution of yolov5 is 3x672x672, but why in https://github.com/NVIDIA-AI-IOT/cuDLA-samples/blob/a2d645b61920fead0cf70c79506518b0a159463c/src/matx_reformat/matx_reformat.cu#L146, the shape of mInput1 tensor was allocated as (1x16x672x672). 
anybody known why???

zerollzeng, 2024-01-27:
Because DLA takes chw16 inputs, which means we need to pad the input to NC/16HW16 first.

dattv, 2024-01-28:
> Because DLA takes chw16 inputs, which means we need to pad the input to NC/16HW16 first.

@zerollzeng  Thank you for your feedback!, 
however there are some extra information I want to inquires, 
- Did you mean 16 in chw16 is 16 bit?
- Because the image included 3 channel (RGB) so i thought the mInput1 tensor must be something like 1x16x672x672*sizeof(half)?
- Is there any different between memory size of cuda variables and dla variables?

zerollzeng, 2024-02-01:
No, the 16 means the C channel is padded to 16, So you actually have a 1x16x672x672 tensor, while only the first 3 channels represent the image.

lynettez, 2024-09-02:
closing since no activity for several month, thanks all!

closed 2024-09-02 by lynettez