NVIDIA-AI-IOT/cuDLA-samples #20: build dla patch error

Railcalibur, 2024-01-26:
hello,
I followed the readme to deploy yolo on DLA. However, it occurred error when I applied dla patch and builded trtexec.

Env:

Jetson AGX Orin
JetPack 5.1.2
CUDA 11.4
TensorRT 8.5.2.2

Step:

git apply trtexec-dla-standalone-trtv8.5.patch

warning: samples/common/sampleEngines.cpp has type 100644, expected 100755
warning: samples/common/sampleOptions.cpp has type 100644, expected 100755
warning: samples/common/sampleOptions.h has type 100644, expected 100755

sudo make

../Makefile.config:13: CUDA_INSTALL_DIR variable is not specified, using /usr/local/cuda by default, use CUDA_INSTALL_DIR=<cuda_directory> to change.
../Makefile.config:18: CUDNN_INSTALL_DIR variable is not specified, using /usr/local/cuda by default, use CUDNN_INSTALL_DIR=<cudnn_directory> to change.
../Makefile.config:38: TRT_LIB_DIR is not specified, searching ../../lib, ../../lib, ../lib by default, use TRT_LIB_DIR=<trt_lib_directory> to change.
../Makefile.config:43: PROTOBUF_INSTALL_DIR variable is not specified, use PROTOBUF_INSTALL_DIR=<protobuf_directory> to change. Using default value: /usr/lib/x86_64-linux-gnu
Linking: ../../bin/trtexec_debug
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::ISoftMaxLayer::setAxis(int)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::INetwork::addTranspose(nvdla::ITensor*, nvdla::Dims4)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::INetwork::addSlice(nvdla::ITensor*, nvdla::Weights, nvdla::Weights, nvdla::Weights, nvdla::Weights, nvdla::SliceLayerMode)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::IProfile::setUseSoftMaxOptz(bool)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::IProfile::setCanCompressStructuredSparseWeights(bool)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::IPoolingLayer::setPoolingPaddingInclusionType(nvdla::PoolingPaddingInclusionType)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::INetwork::addResize(nvdla::ITensor*, nvdla::ResizeMode, nvdla::Weights)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::INetwork::addReshape(nvdla::ITensor*, nvdla::Dims4)'
collect2: error: ld returned 1 exit status
make: *** [../Makefile.config:346: ../../bin/trtexec_debug] Error 1

lynettez, 2024-02-01:
Could you try to `find /user -name "*nvdla_compiler.so*"`? If it existed, you may modify the Makefile to link the path.

Railcalibur, 2024-02-04:
> 你能尝试一下`find /user -name "*nvdla_compiler.so*"`吗？如果存在，可以修改Makefile来链接路径。

Thank you for the reply. The .so file was not exist and the issue was resolved after reflashing the device.

closed 2024-02-04 by Railcalibur