NVIDIA-AI-IOT/cuDLA-samples #21: src/cudla_context_standalone.h:30:10: fatal error: nvscibuf.h: No such file or directory

yunyao2011800, 2024-01-30:
when I run the command: " make clean  & make  run ",  An error occurred：

```

**/usr/local/cuda/bin/nvcc -I /usr/local/cuda/include -I ./src/matx_reformat/ -I /usr/include/opencv4/ -I /usr/include/jsoncpp/ -I /usr/include -gencode arch=compute_87,code=sm_87 -c -o build/decode_nms.o src/decode_nms.cu
g++ -I /usr/local/cuda/include -I ./src/matx_reformat/ -I /usr/include/opencv4/ -I /usr/include/jsoncpp/ -I /usr/include --std=c++14 -Wno-deprecated-declarations -Wall -O2 -c -o build/validate_coco.o src/validate_coco.cpp
g++ -I /usr/local/cuda/include -I ./src/matx_reformat/ -I /usr/include/opencv4/ -I /usr/include/jsoncpp/ -I /usr/include --std=c++14 -Wno-deprecated-declarations -Wall -O2 -c -o build/yolov5.o src/yolov5.cpp
g++ -I /usr/local/cuda/include -I ./src/matx_reformat/ -I /usr/include/opencv4/ -I /usr/include/jsoncpp/ -I /usr/include --std=c++14 -Wno-deprecated-declarations -Wall -O2 -c -o build/cudla_context_hybrid.o src/cudla_context_hybrid.cpp
g++ -I /usr/local/cuda/include -I ./src/matx_reformat/ -I /usr/include/opencv4/ -I /usr/include/jsoncpp/ -I /usr/include --std=c++14 -Wno-deprecated-declarations -Wall -O2 -c -o build/cudla_context_standalone.o src/cudla_context_standalone.cpp
In file included from src/cudla_context_standalone.cpp:31:
src/cudla_context_standalone.h:30:10: fatal error: nvscibuf.h: No such file or directory
   30 | #include "nvscibuf.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:69: build/cudla_context_standalone.o] Error 1**
```

How to solve this problem ？

ou525, 2024-01-31:
#12

yunyao2011800, 2024-01-31:
@ou525    Thank you, I have resolved the issue.

https://developer.nvidia.com/docs/drive/drive-os/archives/6.0.3/linux/sdk/api_reference/nvscibuf_8h_source.html

closed 2024-01-31 by yunyao2011800