NVIDIA-AI-IOT/cuDLA-samples #24: how to test inference time

ou525, 2024-02-02:
i try use chrono,but when i print the time, get 0。
[image]

ou525, 2024-02-02:
> ```
>     auto start0 = std::chrono::system_clock::now();
>     cv::Mat one_img = cv::imread(image_path);
>     bgr_imgs.push_back(one_img);
>     std::vector<cv::Mat> nchwMats = yolov5_infer.preProcess4Validate(bgr_imgs);
> 
>     yolov5_infer.infer();
>     results = yolov5_infer.postProcess4Validation(0.25, 0.5f);
>     printf("Num object detect: %ld\n", results.size());
>     for (auto &item : results)
>     {
>         // left, top, right, bottom, label, confident
>         cv::rectangle(one_img, cv::Point(item[0], item[1]), cv::Point(item[2], item[3]), cv::Scalar(0, 255, 0), 2, 16);
>     }
>     auto end0 = std::chrono::system_clock::now();
>     std::cout << "detect image time: " << std::chrono::duration_cast<std::chrono::milliseconds>(end0- start0).count() << "ms" << std::endl;
> ```

I modified the code with reference to this and added this code to multiple loops. I found that the printing time is getting longer and longer. I don’t know where the problem is.
		for(int k=0;k< 1;k++)
		{
			auto start0 = std::chrono::steady_clock::now();
			cv::Mat one_img = cv::imread(image_path);
			bgr_imgs.push_back(one_img);
			std::vector<cv::Mat> nchwMats = yolov5_infer.preProcess4Validate(bgr_imgs);
			
			yolov5_infer.infer();
			results = yolov5_infer.postProcess4Validation(0.25, 0.5f);
			printf("Num object detect: %ld\n", results.size());
			for (auto &item : results)
			{
				// left, top, right, bottom, label, confident
				cv::rectangle(one_img, cv::Point(item[0], item[1]), cv::Point(item[2], item[3]), cv::Scalar(0, 255, 0), 2,
							  16);
			}
			auto end0 = std::chrono::steady_clock::now();
			std::cout << "detect time:" << std::chrono::duration_cast<std::chrono::milliseconds>(end0- start0).count() << " " << std::chrono::duration_cast<std::chrono::milliseconds>(start0.time_since_epoch()).count() << " " << std::chrono::duration_cast<std::chrono::milliseconds>(end0.time_since_epoch()).count() << std::endl;
		}

ou525, 2024-02-02:
After testing, it was found that it was caused by this function preProcess4Validate. Can you help me take a look? @zerollzeng

yunyao2011800, 2024-02-04:
> n preProcess4Validate. Can you help me take a loo

bgr_imgs.clear();

ou525, 2024-02-04:
@yunyao2011800 Thank you; it was precisely due to this issue, and it has been resolved.

closed 2024-02-04 by ou525