NVIDIA-AI-IOT/cuDLA-samples #28: How to obtain the min and max values of activation and weights？

CangHaiQingYue, 2024-03-05:
I want to deploy a qat model on the chip, but the chip requires min max values for activation and weights. What should I do?
The format of the JSON file is as follows. In fact, this is the format of AIMET. I found that mAP of this Repositorie is better than AIMET.
{
    "activation_encodings": {
        "115": [
            {
                "bitwidth": 8,
                "dtype": "int",
                "is_symmetric": "False",
                "max": 9.063658714294434,
                "min": 0.0,
                "offset": 0,
                "scale": 0.03554375842213631
            }
        ],
 "param_encodings": {
        "conv1.weight": [
            {
                "bitwidth": 8,
                "dtype": "int",
                "is_symmetric": "True",
                "max": 0.08543740957975388,
                "min": -0.08611015230417252,
                "offset": -128,
                "scale": 0.0006727355648763478
            },

lynettez, 2024-03-25:
what chip do you use? is it a common question for TensorRT? could you new an issue in https://github.com/NVIDIA/TensorRT/issues?

lynettez, 2024-09-02:
closing since no activity for several months, thanks!

closed 2024-09-02 by lynettez