NVIDIA-AI-IOT/cuDLA-samples #30: No module named quantization

zyitom, 2024-03-13:
[image]
I've following the instructions and download the pytorch-quantization via '''pip install pytorch-quantization --extra-index-url https://pypi.ngc.nvidia.com''' this link but when i run the qat.py it met the error and said no modules found ,what shoule i do

liuanqi-libra7, 2024-04-24:
@zyitom 
It seems that the library installation was not successful. You can try using the following command to install it.
```shell
$ git clone -b v8.6.1 https://github.com/NVIDIA/TensorRT.git
$ cd TensorRT/tools/pytorch-quantization
$ python setup.py install
```

lynettez, 2024-09-02:
closing since no activity for several months, thanks all!

closed 2024-09-02 by lynettez