NVIDIA-AI-IOT/cuDLA-samples #33: Size of output tensor in ReformatImage function

mayulin0206, 2024-04-22:
[        matx::tensor_t<matx::matxFp16, 5> mHead1Input1 = matx::make_tensor<matx::matxFp16>((matx::matxFp16*)src[0], {1, 16, 84, 84, 16});
        matx::tensor_t<matx::matxFp16, 5> mHead1Output1 = matx::make_tensor<matx::matxFp16>((matx::matxFp16*)mTemp, {1, 16, 16, 84, 84});](url)

The output shape is {1, 16, 84, 84, 16}
why the shape is {1, 16, 84, 84, 16}？ other shape is okey? like{1, 8, 168,84, 16}

Sunhaihua1, 2025-03-07:
Hi, I have the same question, do you have the answer？

lynettez, 2025-09-19:
Sorry for the late reply.

Input tensor shape: {1, 16, 84, 84, 16} (5 dimensions)
Dimension 0: 1 (batch size)
Dimension 1: 16 (first channel group)
Dimension 2: 84 (height)
Dimension 3: 84 (width)
Dimension 4: 16 (second channel group)

Permute operation: .Permute({0, 1, 4, 2, 3})
This reorders the dimensions as follows:
New dim 0 ← Old dim 0: 1
New dim 1 ← Old dim 1: 16
New dim 2 ← Old dim 4: 16 (moved from last position)
New dim 3 ← Old dim 2: 84 (height moved to position 3)
New dim 4 ← Old dim 3: 84 (width moved to position 4)
Output tensor shape: {1, 16, 16, 84, 84}

This operation is channel reorganization - it's moving the second channel dimension (originally at position 4) to position 2, while shifting the spatial dimensions (height/width) toward the end. 
The input has channels split across two dimensions: [batch, ch1, H, W, ch2]
The output reorganizes to: [batch, ch1, ch2, H, W]