NVIDIA-AI-IOT/cuDLA-samples #43: Failed to export on AGX orin 64gb

berkay-karlik, 2024-12-09:
Hello, I'm trying to export an DLA engine on AGX orin. 
I first tried to fine-tune the pt model and export to onnx and cloud and then convert it to engine file on AGX orin, well that failed with errow below. 
Then I tried fetch the existing onnx file on `/data/model` with git lfs but that failed with the same error too. Any guidance would be really helpful. thank you.

```bash
jensen@ubuntu:~/cuDLA-samples/data/model$ bash build_dla_standalone_loadable.sh 
Build DLA loadable for fp16 and int8
&&&& RUNNING TensorRT.trtexec [TensorRT v100300] # /usr/src/tensorrt/bin/trtexec --onnx=data/model/yolov5s_trimmed_reshape_tranpose.onnx --verbose --fp16 --saveEngine=data/loadable/yolov5.fp16.fp16chw16in.fp16chw16out.standalone.bin --inputIOFormats=fp16:chw16 --outputIOFormats=fp16:chw16 --buildDLAStandalone --useDLACore=0
[12/09/2024-12:20:42] [I] === Model Options ===
[12/09/2024-12:20:42] [I] Format: ONNX
[12/09/2024-12:20:42] [I] Model: data/model/yolov5s_trimmed_reshape_tranpose.onnx
[12/09/2024-12:20:42] [I] Output:
[12/09/2024-12:20:42] [I] === Build Options ===
[12/09/2024-12:20:42] [I] Memory Pools: workspace: default, dlaSRAM: default, dlaLocalDRAM: default, dlaGlobalDRAM: default, tacticSharedMem: default
[12/09/2024-12:20:42] [I] avgTiming: 8
[12/09/2024-12:20:42] [I] Precision: FP32+FP16
[12/09/2024-12:20:42] [I] LayerPrecisions: 
[12/09/2024-12:20:42] [I] Layer Device Types: 
[12/09/2024-12:20:42] [I] Calibration: 
[12/09/2024-12:20:42] [I] Refit: Disabled
[12/09/2024-12:20:42] [I] Strip weights: Disabled
[12/09/2024-12:20:42] [I] Version Compatible: Disabled
[12/09/2024-12:20:42] [I] ONNX Plugin InstanceNorm: Disabled
[12/09/2024-12:20:42] [I] TensorRT runtime: full
[12/09/2024-12:20:42] [I] Lean DLL Path: 
[12/09/2024-12:20:42] [I] Tempfile Controls: { in_memory: allow, temporary: allow }
[12/09/2024-12:20:42] [I] Exclude Lean Runtime: Disabled
[12/09/2024-12:20:42] [I] Sparsity: Disabled
[12/09/2024-12:20:42] [I] Safe mode: Disabled
[12/09/2024-12:20:42] [I] Build DLA standalone loadable: Enabled
[12/09/2024-12:20:42] [I] Allow GPU fallback for DLA: Disabled
[12/09/2024-12:20:42] [I] DirectIO mode: Disabled
[12/09/2024-12:20:42] [I] Restricted mode: Disabled
[12/09/2024-12:20:42] [I] Skip inference: Enabled
[12/09/2024-12:20:42] [I] Save engine: data/loadable/yolov5.fp16.fp16chw16in.fp16chw16out.standalone.bin
[12/09/2024-12:20:42] [I] Load engine: 
[12/09/2024-12:20:42] [I] Profiling verbosity: 0
[12/09/2024-12:20:42] [I] Tactic sources: Using default tactic sources
[12/09/2024-12:20:42] [I] timingCacheMode: local
[12/09/2024-12:20:42] [I] timingCacheFile: 
[12/09/2024-12:20:42] [I] Enable Compilation Cache: Enabled
[12/09/2024-12:20:42] [I] errorOnTimingCacheMiss: Disabled
[12/09/2024-12:20:42] [I] Preview Features: Use default preview flags.
[12/09/2024-12:20:42] [I] MaxAuxStreams: -1
[12/09/2024-12:20:42] [I] BuilderOptimizationLevel: -1
[12/09/2024-12:20:42] [I] Calibration Profile Index: 0
[12/09/2024-12:20:42] [I] Weight Streaming: Disabled
[12/09/2024-12:20:42] [I] Runtime Platform: Same As Build
[12/09/2024-12:20:42] [I] Debug Tensors: 
[12/09/2024-12:20:42] [I] Input(s): fp16:+chw16
[12/09/2024-12:20:42] [I] Output(s): fp16:+chw16
[12/09/2024-12:20:42] [I] Input build shapes: model
[12/09/2024-12:20:42] [I] Input calibration shapes: model
[12/09/2024-12:20:42] [I] === System Options ===
[12/09/2024-12:20:42] [I] Device: 0
[12/09/2024-12:20:42] [I] DLACore: 0
[12/09/2024-12:20:42] [I] Plugins:
[12/09/2024-12:20:42] [I] setPluginsToSerialize:
[12/09/2024-12:20:42] [I] dynamicPlugins:
[12/09/2024-12:20:42] [I] ignoreParsedPluginLibs: 0
[12/09/2024-12:20:42] [I] 
[12/09/2024-12:20:42] [I] === Inference Options ===
[12/09/2024-12:20:42] [I] Batch: Explicit
[12/09/2024-12:20:42] [I] Input inference shapes: model
[12/09/2024-12:20:42] [I] Iterations: 10
[12/09/2024-12:20:42] [I] Duration: 3s (+ 200ms warm up)
[12/09/2024-12:20:42] [I] Sleep time: 0ms
[12/09/2024-12:20:42] [I] Idle time: 0ms
[12/09/2024-12:20:42] [I] Inference Streams: 1
[12/09/2024-12:20:42] [I] ExposeDMA: Disabled
[... 195 lines cut ...]
[12/09/2024-12:20:46] [I] Selected Device UUID: GPU-22323d72-1ce1-5535-9602-51e3f8dcb6b8
[12/09/2024-12:20:46] [I] Compute Capability: 8.7
[12/09/2024-12:20:46] [I] SMs: 6
[12/09/2024-12:20:46] [I] Device Global Memory: 62840 MiB
[12/09/2024-12:20:46] [I] Shared Memory per SM: 164 KiB
[12/09/2024-12:20:46] [I] Memory Bus Width: 256 bits (ECC disabled)
[12/09/2024-12:20:46] [I] Application Compute Clock Rate: 1.3 GHz
[12/09/2024-12:20:46] [I] Application Memory Clock Rate: 0.408 GHz
[12/09/2024-12:20:46] [I] 
[12/09/2024-12:20:46] [I] Note: The application clock rates do not reflect the actual clock rates that the GPU is currently running at.
[12/09/2024-12:20:46] [I] 
[12/09/2024-12:20:46] [I] TensorRT version: 10.3.0
[12/09/2024-12:20:46] [I] Loading standard plugins
[12/09/2024-12:20:46] [I] [TRT] [MemUsageChange] Init CUDA: CPU +2, GPU +0, now: CPU 31, GPU 3249 (MiB)
[12/09/2024-12:20:49] [I] [TRT] [MemUsageChange] Init builder kernel library: CPU +928, GPU +754, now: CPU 1002, GPU 4048 (MiB)
[12/09/2024-12:20:49] [I] Start parsing network model.
Could not open file data/model/yolov5_trimmed_qat.onnx
[12/09/2024-12:20:49] [E] [TRT] ModelImporter.cpp:914: Failed to parse ONNX model from file: data/model/yolov5_trimmed_qat.onnx!
[12/09/2024-12:20:49] [E] Failed to parse onnx file
[12/09/2024-12:20:49] [I] Finished parsing network model. Parse time: 0.00242894
[12/09/2024-12:20:49] [E] Parsing model failed
[12/09/2024-12:20:49] [E] Failed to create engine from model or file.
[12/09/2024-12:20:49] [E] Engine set up failed
&&&& FAILED TensorRT.trtexec [TensorRT v100300] # /usr/src/tensorrt/bin/trtexec --minShapes=images:1x3x672x672 --maxShapes=images:1x3x672x672 --optShapes=images:1x3x672x672 --shapes=images:1x3x672x672 --onnx=data/model/yolov5_trimmed_qat.onnx --useDLACore=0 --buildDLAStandalone --saveEngine=data/loadable/yolov5.int8.int8hwc4in.fp16chw16out.standalone.bin --inputIOFormats=int8:dla_hwc4 --outputIOFormats=fp16:chw16 --int8 --fp16 --calib=data/model/qat2ptq.cache --precisionConstraints=obey --layerPrecisions=/model.24/m.0/Conv:fp16,/model.24/m.1/Conv:fp16,/model.24/m.2/Conv:fp16
```

wangyi2019py, 2024-12-13:
Did you debug it?

berkay-karlik, 2024-12-13:
no I used another [repository](https://github.com/marcoslucianops/DeepStream-Yolo/blob/master/docs/YOLOv5.md)

lynettez, 2025-09-19:
Sorry for the late reply. It seems the ONNX files were not pulled correctly. Please try running `lfs pull` again.