NVIDIA-AI-IOT/cuDLA-samples #44: Output precision diff between GPU and DLA

paperybox, 2025-02-05:
Hi Nvidia Developer,
    I tried QAT training, following the https://github.com/NVIDIA-AI-IOT/cuDLA-samples/blob/main/export/README.md.  I found the model infer output result do not match between DLA and GPU, Is this in expectation?

Trainning Device: RTX3060-12G
Infer Device: Jetson AGX Orin 64G

paperybox, 2025-02-06:
Hi Nvidia Developer,
           I tried to use PTQ on my model with nvinfer1::IInt8EntropyCalibrator2, but the results seem to have some problems with DLA PTQ., Why does DLA PTQ int8 accuracy drop and accuracy stay flat？

Infer Device: Jetson AGX Orin 64G, JetPack 5.1.4 Linux , TRT-8502-EntropyCalibration2 
| Device            | DataType  | Precision | Recall  | Latency  |
|-------------------|-----------|-----------|---------|----------|
| OrinX[GPU] | fp32      | 91.01%    | 95.01%  | 49.5195  |
|                   | fp16      | 91.01%    | 95.01%  | 22.1503  |
|                   | Force int8    | 90.70%    | 92.47%  | 13.4525  |
|                   | PTQ int8  | 90.81%    | 94.86%  | 13.4464  |
| OrinX[DLA] | Force int8| 90.47%    | 92.36%  | 19.2831  |
|                   | PTQ int8  | 68.16%    | 92.65%  | 19.2722  |

lynettez, 2025-09-19:
> I found the model infer output result do not match between DLA and GPU, Is this in expectation?
Yes, it’s expected that the DLA outputs are not bit-aligned with the GPU, since the internal computation differs between the two hardware platforms. 
How is the QAT accuracy (mAP) on DLA? 

Using PTQ with an INT8 Calibrator is not recommended, as it may not achieve good accuracy for certain models.