NVIDIA-AI-IOT/cuDLA-samples #45: fatal error: __config: No such file or directory while build_matx_reformat

fangbaolei, 2025-04-11:
bash src/matx_reformat/build_matx_reformat.sh

/usr/local/cuda/targets/aarch64-linux/include/nvtx3/nvToolsExt.h:179:2: error: #error "Trying to #include NVTX version 3 in a source file where an older NVTX version has already been included.  If you are not directly using NVTX (the NVIDIA Tools Extension library), you are getting this error because libraries you are using have included different versions of NVTX.  Suggested solutions are: (1) reorder #includes so the newest NVTX version is included first, (2) avoid using the conflicting libraries in the same .c/.cpp file, or (3) update the library using the older NVTX version to use the newer version instead."
  179 | #error "Trying to #include NVTX version 3 in a source file where an older NVTX version has already been included.  If you are not directly using NVTX (the NVIDIA Tools Extension library), you are getting this error because libraries you are using have included different versions of NVTX.  Suggested solutions are: (1) reorder #includes so the newest NVTX version is included first, (2) avoid using the conflicting libraries in the same .c/.cpp file, or (3) update the library using the older NVTX version to use the newer version instead."
      |  ^~~~~
In file included from /usr/local/cuda/targets/aarch64-linux/include/cuda/std/optional:25,
                 from /usr/local/cuda/targets/aarch64-linux/include/cub/detail/nvtx.cuh:52,
                 from /usr/local/cuda/targets/aarch64-linux/include/cub/device/device_adjacent_difference.cuh:41,
                 from /usr/local/cuda/targets/aarch64-linux/include/cub/cub.cuh:62,
                 from /home/weilan/work/cuDLA-samples/src/matx_reformat/MatX/include/matx/transforms/cub.h:39,
                 from /home/weilan/work/cuDLA-samples/src/matx_reformat/MatX/include/matx/transforms/reduce.h:44,
                 from /home/weilan/work/cuDLA-samples/src/matx_reformat/MatX/include/matx/transforms/cgsolve.h:35,
                 from /home/weilan/work/cuDLA-samples/src/matx_reformat/MatX/include/matx/transforms/transforms.h:36,
                 from /home/weilan/work/cuDLA-samples/src/matx_reformat/MatX/include/matx.h:51,
                 from /home/weilan/work/cuDLA-samples/src/matx_reformat/matx_reformat.cu:25:
/home/weilan/work/cuDLA-samples/src/matx_reformat/build/_deps/libcudacxx-src/include/cuda/std/detail/libcxx/include/optional:149:10: fatal error: __config: No such file or directory
  149 | #include <__config>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/matx_reformat.dir/build.make:77: CMakeFiles/matx_reformat.dir/matx_reformat.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:117: CMakeFiles/matx_reformat.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

env:
[image]

capyun, 2025-07-25:
I get this error too, anyone can help me?

lynettez, 2025-09-19:
Sorry for the late reply, we didn't test with cuda 12.6. Did you resolve this on your end? we'll have a try to see if it can be reproduced.